CCHS-Melbourne/3D-Printers #2: CR 10: OctoPrint Server randomly resets

PancakeLegend, 2021-01-05:
The OctoPrint Server on the CR 10 is a Pi 1 which should be replaced with a newer (more reliable) Pi.

PancakeLegend, 2021-01-06:
Check that the power circuit isn't the culprit. Turning the light on and off can cause printers to power-cycle.

jubeormk1, 2021-04-21:
> Check that the power circuit isn't the culprit. Turning the light on and off can cause printers to power-cycle.

What power circuit are refering to? I added a 60W power supply to the 3d printing cluster to power the raspberry pies. The pi cr10.local keeps reporting low voltage. I suspect that it has to do with the CR10 control board draining power from the raspberry pi usb port

kerriwait, 2021-04-21:
@jubeormk1 we need to modify the USB cables for the pi to control board links to remove the 5V connection. This will stop the control boards from trying to power themselves via the pi.

jubeormk1, 2021-04-29:
Agreed Kerri. I Have some USB A connector that we can use to interrupt neatly the Vcc line, worth trying and labeling the cable since it can result confusing for other applications.

brainstorm, 2022-02-05:
The TAZ RPi constantly complains about low voltage, we need better power sources than crappy wallwarts... perhaps just break out each of the 3D printer's PSU 5V rails and connect it to the PI? That'll just cut the power to the PIs when the printer is not in use, which can be a good side effect, IMO... I'd perhaps throw one of those UPSs to the mix: https://www.aliexpress.com/item/32969695165.html

PancakeLegend, 2022-02-05:
This has been shown to be a Raspberry Pi issue. Even with a suitable power supply with sufficient current, the undervoltage flag will still be triggered during boot.
On my own setup with a buck converter directly off the power supply, the undervolt issue still occurred until I increased the voltage to 5.2v on the buck converter. This is still within the ±2% Raspberry Pi specification.

brainstorm, 2022-02-05:
> This has been shown to be a Raspberry Pi issue. Even with a suitable power supply with sufficient current, the undervoltage flag will still be triggered during boot. On my own setup with a buck [...]

Gotcha, good to know, we can then put those cheap sub-1$ voltage regulators, it makes sense because otherwise Octoprint gets fairly intrusive/annoying with warning popups on the interface and right before printing.

PancakeLegend, 2022-02-05:
@brainstorm Worth noting that turning on the lights connected to that power point is still an issue. There is a  'No power tools' sign at that power point. Clearly, this has been a known issue for a long time. A UPS for everything on that rack has been discussed, though maybe an electrician should be consulted.