CCHS-Melbourne/3D-Printers #4: Taz: Cold end fan requires bracket

PancakeLegend, 2021-04-07:


jubeormk1, 2021-04-21:
Late report: Last Tuesday the 13th of April we "crafted" a temporary metal sheet fan bracket.

Next action is to print and install a long lasting ABS bracket.

jubeormk1, 2021-05-04:
Printed a Cali Mouse using a "Custom" cura profile based on the [cchs wiki for taz 5](https://github.com/CCHS-Melbourne/3D-Printers/wiki/%5BPrinter%5D-Taz-5-(LulzBot)) and the loaded white ABS.
- Heated bed is not heating up. See [this issue for more details](https://github.com/CCHS-Melbourne/3D-Printers/issues/6)
- The filament (white one) has plenty of moisture and there is some under extrusion.
- Adhesion of the part is fine.

I would say that we do not need to print a new ABS bracket for the cold rod.

jubeormk1, 2021-05-05:
Shall we close this issue?

PancakeLegend, 2021-06-30:
New fan shroud appears to be working.

closed 2021-06-30 by PancakeLegend