CCHS-Melbourne/3D-Printers #5: Tiertime UP! 2 3d printer refurbishment

jubeormk1, 2021-04-21:
This 3d printed was anonymously donated to the space, needing some caring but in general good conditions. It is a proprietary 3d printed that can mainly be operated using Tiertime software on a windows environment. The current issues and actions follow:

Issues:
1. [CLOSED] Case replacement: The case around the carriage is broken
2. Inconsistent extrusion and filament grinding. @brainstorm  and I suspect that it is due to nozzle clogging
3. The system firmware is not available online. uSD card corruption would lead to an unusable printer

Actions:
1.1 @PancakeLegend has design and printed a new case for the carriage that can be connected to a dry box with an standard tube fitting.
2.1 @brainstorm has provided and installed a new nozzle that need to be validated with a test print

Next steps:
2.2 Test print to close second problem
3.1 Dump data from the printer uSD for backup and reverse engineering

PancakeLegend, 2021-04-21:
Nice one. I've added folders for each printer for backups, profiles, calibration gcode, etc.

brainstorm, 2021-04-27:
2.2 is done by @jubeormk1 or @PancakeLegend, I reckon?
3.1 is done too, see explanation I pushed in: https://github.com/CCHS-Melbourne/3D-Printers/blob/master/Up%20Plus%202/README.md

TL;DR: The internal SD card can get corrupted, the actual 3D printer firmware is in the IC, so it'd make more sense to [USB-Wireshark the protocol](https://wiki.wireshark.org/CaptureSetup/USB#Windows) if we want this printer to work with Octoprint.

closed 2021-04-27 by brainstorm