CCHS-Melbourne/3D-Printers #6: TAZ: Heated bed not heating

jubeormk1, 2021-05-04:
Doing a basic test in the printer looks like the bed heater is not heating. I did some test after disconnecting the modular Anderson connectors:
- I open the connector for the heated bed checked the voltage going to the heating bed and looks fine
- The measured resistance of the heating bed is 1.5 Ohms

Maybe continuity issues on the cable?

brainstorm, 2022-01-29:
Did some thorough testing and concluded that the bed heating element was the one at fault, but it turns out I was wrong? Can you double check these pictures, @jubeormk1 ?

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

brainstorm, 2022-01-29:
Will check the automotive fuse they mention here...: https://forum.lulzbot.com/t/taz-bed-not-heating-up-help-with-troubleshooting/5060/6

brainstorm, 2022-01-29:
Nope, the fuse is fine :/
[image]

[image]

brainstorm, 2022-02-05:
Fixed! It was the bed connector that goes from the "Rambo" board (mainboard of the TAZ) to the bed connector... the internal leads were bent and were not doing proper contact, I managed to print a calibrator cat today, thanks everybody for the suggestions over Discord and elsewhere :)

brainstorm, 2022-02-12:
The bed connector is unreliable, needs to be replaced.

brainstorm, 2022-02-14:
Found the real culprit and fixed it today for good... the convenience of connectors is overrated.

[image]
[image]
[image]

closed 2022-02-14 by brainstorm